arjenhiemstra/ithowifi #334: Air Quality (%) > 100%

VincentSC, 2025-07-20:
## Describe the bug
Related to this bug: https://github.com/jasperslits/haithowifi/issues/66 - there it's solved by just throw an "unknown" when larger than 100%:
```
if isinstance(value, (int, float)) and float(value) > 100:
    _error_description = "Unknown error"
    value = None
```

I do see that more people have > 100%. @dirixmjm and @benjamin-dcs both have exactly that same 119.5% - tagging them here, to learn what model they have.

## To Reproduce
Steps to reproduce the behaviour:
1. Go to Itho Status
2. See that "Air Quality (%)" is larger than 100%

## Device information
- Firmware version:  2.8.0
- Hardware revision: NON-CVE 1
- Type/model number Itho: HRU 350 HR
- CC1101 RF module enabled: yes
- In case of wifi issues, wifi access point make and model: N/A

## Syslog
Please provide the syslog from the Syslog page. If truncated, please also provide the recycled file:
```
025-07-20 00:10:47 I: Hostname: nrg-itho-[REDACTED]
2025-07-20 00:10:47 I: mDNS: started
2025-07-20 00:10:46 I: Webserver: started
2025-07-20 00:10:46 E: MQTT: connection failed, System config: 1
2025-07-20 00:10:44 I: I2C init: QueryStatus
2025-07-20 00:10:44 I: I2C init: QueryStatusFormat - items:33
2025-07-20 00:10:43 I: I2C init: QueryDevicetype - mfr:0x00 type:0x2B fw:0x04 hw:0x0A
2025-07-20 00:10:43 I: Setup: init of CC1101 RF module successful
2025-07-20 00:10:41 I: Timezone: Europe/Amsterdam, specifier CET-1CEST,M3.5.0,M10.5.0/3 
2025-07-20 00:10:41 I: WiFi: connection successful
2025-07-20 00:10:35 I: wifi AP mode started
2025-07-20 00:10:33 E: Unable to set wifi disconnect
2025-07-20 00:10:31 I: I2C sniffer capable hardware: yes
2025-07-20 00:10:31 I: HW rev: NON-CVE 1, FW ver.: 2.8.0
2025-07-20 00:10:30 I: Device UUID: dfc76a20-6b63-45a0-bb17-0b85d908b948
2025-07-20 00:10:30 I: System boot, last reset reason: SW_RESET
2025-07-20 00:10:26 I: Reboot requested
    212386 I: WiFi: connection successful
     15609 I: I2C init: QueryStatus
     15440 I: I2C init: QueryStatusFormat - items:33
     15245 I: I2C init: QueryDevicetype - mfr:0x00 type:0x2B fw:0x04 hw:0x0A
     10808 I: Hostname: nrg-itho-[REDACTED]
     10593 I: mDNS: started
     10399 I: Webserver: started
      8427 I: Timezone: Europe/Amsterdam, specifier CET-1CEST,M3.5.0,M10.5.0/3 
      8258 I: wifi AP mode started
      6507 E: Unable to set wifi disconnect
      4026 I: I2C sniffer capable hardware: yes
      3906 I: HW rev: NON-CVE 1, FW ver.: 2.8.0
      3800 I: Device UUID: dfc76a20-6b63-45a0-bb17-0b85d908b948
      3489 I: System boot, last reset reason: POWERON_RESET
     15649 I: I2C init: QueryStatus
     15456 I: I2C init: QueryStatusFormat - items:75
     15262 I: I2C init: QueryDevicetype - mfr:0x00 type:0x0B fw:0x14 hw:0x26
     10798 I: Hostname: nrg-itho-[REDACTED]
     10646 I: mDNS: started
     10515 I: Webserver: started
      8527 I: Timezone: Europe/Amsterdam, specifier CET-1CEST,M3.5.0,M10.5.0/3 
      8429 I: wifi AP mode started
      6631 E: Unable to set wifi disconnect
      3948 I: I2C sniffer capable hardware: yes
      3769 I: HW rev: NON-CVE 1, FW ver.: 2.8.0
      3597 I: Device UUID: dfc76a20-6b63-45a0-bb17-0b85d908b948
      3424 I: System boot, last reset reason: POWERON_RESET
     10481 I: Hostname: nrg-itho-[REDACTED]
     10226 I: mDNS: started
     10014 I: Webserver: started
      7990 I: Timezone: Europe/Amsterdam, specifier CET-1CEST,M3.5.0,M10.5.0/3 
      7871 I: wifi AP mode started
      6166 E: Unable to set wifi disconnect
      3563 I: I2C sniffer capable hardware: yes
      3442 I: HW rev: NON-CVE 1, FW ver.: 2.8.0
      3345 I: Device UUID: dfc76a20-6b63-45a0-bb17-0b85d908b948
      3277 I: System boot, last reset reason: POWERON_RESET
  ```

VincentSC, 2025-07-20:
My guess that it's HRU350 related seems to have some foundation, I just found

dirixmjm, 2025-07-20:
@VincentSC yes HRU 350. I guess that is also the root cause for the MQTT reported issue.

VincentSC, 2025-07-21:
So the HRU 350 is different again. :)

I got no idea how to fix this, other then sharing debug logs and help testing things out.

jelmerwouters-topicus, 2025-07-21:
> For the record, the DemandFlow reports the same value, but probably it gets this from the HRU350. Maybe the fact it is counted as % is wrong, and rather is a PPM value?

The value is static, it doesn't change over time

VincentSC, 2025-07-21:
Yup, always 119.5. Hence it was so easy to find fellow victims

arjenhiemstra, 2025-07-21:
I currently have none of my itho CO2 sensors connected. Looking at my CVE, the AirQuality (%) reports "not available", it actually also is 0xEF behind the scenes. 

This can be explained as follows, the AirQuality (%) is part of the status items of the HRU350 but also of the 31DA status message. 

The 31DA message of the CVE is parsed here:
https://github.com/arjenhiemstra/ithowifi/blob/81c233c77231243b1ead63c9022fd14e99b99664/software/NRG_itho_wifi/main/IthoSystem.cpp#L1082-L1099

Its the first part of the 31DA message, you can see the check for >100% (ie. >200 in the message) and if that is the case an error message is selected from this list:
https://github.com/arjenhiemstra/ithowifi/blob/81c233c77231243b1ead63c9022fd14e99b99664/software/NRG_itho_wifi/main/devices/error_info_labels.h#L38-L55

You can se 0xEF translates to "not available".

This mechanism has not been implemented for device specific status items. Those do not follow a standard format and to make it possible te implement status items for all devices I left this error message translation code out. The HRU350 happens to have a status label AirQuality (%) as well, just as the 31DA message. It is just not translated to an human readable error message.
It could probably be implemented based on the data format that is also available for all those status items but I never looked into it.

So I think we can safely conclude its a feature of the itho firmware and indicates a missen sensor to base the air quality on. Could wel be a missing or malfunctioning CO2 sensor.

VincentSC, 2025-07-22:
I looked into manuals and product descriptions, and I simply could not find anything.

To me it makes no sense to deliver sucha product without CO2-sensor. But it was also mentioned somewhere that some add-ons that even carry the name CO2, actually missed sich sensor completely.

So, if somebody else could also try to find any mention of CO2 yes/no on a WHR350 compared to other models, that would help confirm the "missing sensor theory"

VincentSC, 2025-07-26:
I connected several devices, but not report CO2. Are there CO2-devices which actually measure that?

The HASS-plugin does not read data from devices, so that's another obstacle

arjenhiemstra, 2025-07-26:
Air quality is not directly measured but always a derivative from other measurements. 
I expect itho to just translate a CO2 measurement, maybe combined with temperature and humidity measurement, to a percentage value.